vertigo235/Prusa-Firmware #2: question: BMG 16 versus 32 einsy MK3S

ruedli, 2019-04-11:
Does the 16 / 32 refer to bits used for microstepping and an BMG? If so, I believe for an MK3 einsy this is usually 32 bits. For the 16, do I need to set the esteps to 415 instead of 830?

vertigo235, 2019-04-15:
Yes this indicates the microstepping, the 16 version is good for folks using a .9 degree stepper motor.  It helps keep the step rate down as to not overload the 8bit board.  

If you're using a 1.8 degree stepper motor, the 32 microstep build is probably fine.

With Prusa firmware you do not need to change the ESTEPS when adjusting the microstepping.

closed 2019-04-15 by vertigo235

Photogad, 2021-02-11:
@vertigo235 I notice there is no longer the  32 variant in the release files, only 16 variant.

Does 16 variant used for both 0.9 and 1.8 stepper now?

Im using LDO 0.9 extruder motor myself and not sure what to do

Photogad, 2021-02-11:
Check here
https://github.com/bgiot/Prusa-Firmware-BMG

He suggest actually 

`#define TMC2130_USTEPS_E 8`

For 0.9 motor and not 16

So im not sure who is correct

ruedli, 2021-02-16:
With the gear reduction 3:1 the number of steps goes beyond the
capabilities of the Einsy board when you need to send 32 pulses for one
motor step. Hence 16 was recommended and with 32 you run into problems with
fast retracts (as e.g. for the MMU filament change). So giving up 32 is
less confusing.

The same is true for 0.9 versus 1.8 degree: the 0.9 degre needs twice as
many steps for the same movement, Hence also here 8 is preferred over 16,
so it seems.

Note that whatever you burn in the firmware as esteps, if the movement is
not correct, you can set it afterwards with an M350 and M92:

M350 - Set microstepping mode M350: Set microstepping mode
<https://reprap.org/wiki/G-code#M350:_Set_microstepping_mode>

After this you should probably also set your number of steps per unit with
an M92:

M92 Set Axis steps-per-unit M92: Set axis_steps_per_unit
<https://reprap.org/wiki/G-code#M92:_Set_axis_steps_per_unit>

Then store these to EEPROM with an M500

vertigo235, 2021-02-16:
@Photogad The variant only sets the default, which is really only relevant if you reset your EEPROM, you should be setting the esteps and microstepping using GCODE and saving it to EEPROM anyhow.  You can choose whatever you want, when I started this you couldn't set the microstepping VIA GCODE, but you can now so there is really no need to have seperate variants for microstep settings.