vertigo235/Prusa-Firmware #6: Variant MK3S-Bondtech-16-SliceHT-SilverPSU?

lordnibbler22, 2019-11-18:
Hello vertigo235!

Can you add another variant?

I am using the Bondtech Mosquito Upgrade on my MK3S Full Bear.

In 1_75mm_MK3S-Bondtech-16-SliceHT-EINSy10a-E3Dv6full.h the DeltaElectronics PSU is uncomment. But i have a SilverPSU installed.

thx!

vertigo235, 2020-02-08:
Will do, as an FYI you don't actually have to use the silver PSU, but it's better because it doesn't turn off your z motors when idle.

Photogad, 2020-04-21:
What is SilverPSU? My PSU is silver came with mk3 in 2018 but I doubt its better than the newer PSU?

vertigo235, 2020-04-21:
@Photogad The Silver PSU is older, but it's fine, the SilverPSU build leaves the Z motors turned on when the printer is idle, which is actually perferable if you don't want the chance of your X axis becoming unlevel while idle.

For the newer Delta PSU Prusa changed all their builds to disable the Z motors because the Delta PSU makes some sort of undesirable noise while idle if the motors are not turned off.

Photogad, 2020-04-21:
> @Photogad The Silver PSU is older, but it's fine, the SilverPSU build leaves the Z motors turned on when the printer is idle, which is actually perferable if you don't want the chance of your X axis [...]

Ive noticed after installing the mmu2s, my heat bed light flickers and sometimes goes off for a second when the bed should be staying heated. I wonder if its too much power for the silver old psu to handle?

vertigo235, 2020-04-24:
@Photogad maybe try a PID tune?

vertigo235, 2020-04-28:
I added this to the latest builds, closing this issue.

closed 2020-04-28 by vertigo235