vertigo235/Prusa-Firmware #7: Bondtech 3.8.1 updates

3d-gussner, 2019-11-19:
Hi @vertigo235,

don't know if you saw that already but Bondtech changed few things in their Prusa Firmware, see https://github.com/BondtechAB/Bondtech_Prusa_i3/commit/37b7033617dde62531d7438f67b56f5c39bfb9bb

As the they didn't make a fork of the Prusa Firmware and have the source code in a zip file, I made a new branch https://github.com/Zaribo/Zaribo-Firmware/tree/Bondtech_Prusa-Firmware to make it bit easier to see the changes they made in the Firmware.

vertigo235, 2020-02-08:
Thanks @3d-gussner !  Yes I need to do this, they have the load more distances adjusted but this is only for MMU2S.  Maybe I'll jump on this today, I'm honored that you like the script and are looking into using drone, thank you so much for your work on the pf-build.sh script, that's what made this so easy to do.

3d-gussner, 2020-04-11:
I close this issue as you read it.

vertigo235, 2020-05-21:
I have made these changes, but I have no way to test

Photogad, 2021-02-11:
@vertigo235 

I think the mosquito bondtech variant needs some more changes merged in. See comparison of yours and official bondtech  mosquito file below

[image]