vertigo235/Prusa-Firmware #8: Zaribo OLED Firmware

LupoLoopy, 2020-05-02:
Hi

I appreciate this adds a whole new set of variants to your project, but figured it would make sense to request it.

https://zaribo.com/home/245-oled-display-20x4-white-black-by-ldo.html

I'm interested running the LDO OLED board (as sold by Zaribo) on my Prusa, but I am currently running other stuff which requires FW tweaks (BMG, Silver PSU etc). 

I use your firmware repository right now to make that easy (it's great! Being able to try the RCs devoid of effort compiling is also nice!). I was wondering if you'd consider adding an OLED SKU to your releases?

vertigo235, 2020-05-21:
I've made these changes and I do not think a new OLED variant is required, I'm running the OLED build on my stock printer. @3d-gussner is there any reason to have a special build for the OLED screen?

3d-gussner, 2020-05-21:
With fw390 we had some changes for oled Display compared to fw381. Please check the Zaribo github what has changed. Otherwise I post tomorrow the changes.

vertigo235, 2020-05-21:
I cherry picked these changes from the Zaribo FW repo and they appear to work fine

https://github.com/vertigo235/Prusa-Firmware/commit/0b8848a4a4474f84bee5c9202bc1805672632cc4

I see this additional commit, I guess you are recommending we include this change as well?

https://github.com/Zaribo/Zaribo-Firmware/commit/9394494e586760e04490eeaab392e136adbf07f7

3d-gussner, 2020-05-21:
I will compile tomorrow your firmware and report back with the oled.