vertigo235/Prusa-Firmware #14: [BUG]Breaks OctoPrint Mesh bed plugins

Smithc9494, 2020-08-18:
**Printer type** - [MK3S]
**Printer firmware version**-  [3.9.1 (Most Recent)

**MMU Upgrade** - [N/A]
**MMU upgrade firmware version [N/A]

**SD card or USB/Octoprint**
 Both

**Describe the bug**
After upgrading to this version I can no longer run a Mesh Bed test with any of the plugins for OctoPrint. They will run the G80 N3 and G81 commands but the plugin never returns the results. You can see them run and return results in the Octoprint terminal but the plugin itself will not show the results.   I tested with the stock Prusa firmware and it works fine. A couple things on my specific printer to be aware of although I don't think they affect this. I have the filament sensor turned off as well as crash detection.  I'm using 7x7 Mesh bed leveling with magnet compensation. 

**To Reproduce**
  Run a mesh bed test with either Prusa Leveling Guide or Prusa Mesh Leveling Octoprint Plugins.
  
**Expected behavior**
  The bed test should kick off and return the results in the Octoprint plugin
  
**G-code**
G28 W ; home all without mesh bed level
M400
G80 N3; mesh bed leveling
G81 ; check mesh leveling results

**Video**
  N/A

audiorazor, 2020-09-04:
Most of the mesh bed plugins that are still being maintained have updated and should work now.   Its not a bug in 3.9, they just changed the way the information was being used or displayed.

vertigo235, 2020-09-27:
This is related to a change in how the Prusa master is reporting the mesh now.  From Scott Rini who is the developer of one of the mesh plugins. 

"n 3.9.1, the output of g81 changed slightly and now outputs carriage return & newline instead of just newline"

closed 2020-09-27 by vertigo235

scottrini, 2020-09-27:
My plug in was fixed in release [1.0.9](https://github.com/scottrini/OctoPrint-PrusaLevelingGuide/releases/tag/1.0.9).
I think PrusaMeshMap still needs a fix, but I believe that plug in is no longer maintained. If necessary, I can fork and fix that one as well, it's a minor change.