vertigo235/Prusa-Firmware #17: [ENHANCEMENT] Check new PF-build.sh PR2850

3d-gussner, 2020-11-02:
I am preparing a new version of `PF-build.sh` where the arguments are replaced by options/flags.
As you are using the `PF-build.sh` to build your custom firmware, I would like you to test the changes and report back in the PR I prepared.
https://github.com/prusa3d/Prusa-Firmware/pull/2850

As soon I get feedback and it is not anymore WIP, I will also update the README.md

vertigo235, 2020-11-06:
Thanks @3d-gussner , I'll see if I can test it soon and let you know.

3d-gussner, 2020-11-14:
@vertigo235 Any update?

3d-gussner, 2020-12-07:
@vertigo235 Did you have time to test it?
I would like to change the WIP state to regular PR, just waiting for some feedback.

vertigo235, 2020-12-15:
Hey @3d-gussner , I'm sorry but I have not had a chance to test it, been busy with other stuff.  Please don't hold up on my account,

3d-gussner, 2021-06-02:
Closing this issue as you have updated it lately for FW3.10.0

closed 2021-06-02 by 3d-gussner