vertigo235/Prusa-Firmware #18: [FEATURE REQUEST]3.9.3 feature review as merge candidate to community versions.

ruedli, 2021-01-26:
MK3S+ BMG extruder 3.9.2
MMU2S 1.06

Prusa's 3.9.3 lists SuperPinda support for MK2.5 and more precise XYZ calibration (making use of the SuperPinda). Is a 3.9.3 merge useful for the community versions?
  
A merge (or statement in the release notes) to which extend 3.9.3 features are in the community firmware binaries of 3.9.2 would be useful.

Photogad, 2021-02-06:
I'd like to see a 3.9.3 release too, however I am patient, will wait for vertigo to come out for more amazing firmware :)

ruedli, 2021-02-06:
I think it only affects mini and 2.5 users, so personally I can wait. Only
missing on more detailed calibration measurements, but with a bear frame
the skew is neglectable.

ruedli, 2021-05-08:
We are all in for 3.10 I  guess ;-)