vertigo235/Prusa-Firmware #21: [QUESTION]<Missing file from Source code Version 3.10>

iankds05, 2021-11-23:
I have downloaded the Source code for Version 3.10 Prusa-Firmware, but when compiling with Arduino IDE the error indicates that the file "Configuration_prusa.h" is missing. Can this be rectified and included in the Source Zip file and then uploaded, as I have made changes and wish to make my own HEX file

domhardt, 2021-11-23:
Using VS Code instead of Arduino IDE is highly recommended.

iankds05, 2021-11-23:
I have used PlatformIo with Atom and VScode and although there is a successful compile the hex file produced just produces rubbish on the MK3s LCD screen and the buzzer just keep complaining

vertigo235, 2022-01-21:
I recommend compiling with the PF-build.sh script, this is how the releases are created.

closed 2022-01-21 by vertigo235